ros2-java/ros2_java #160: ros_android compile issue, on dashing branch with colcon build. Running UBUNTU 18.04 and ROS Dashing

ahmadnav, 2020-12-14:
ll the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)
--- stderr: connext_cmake_module                                                                         
CMake Error at cmake/Modules/FindConnext.cmake:236 (list):
  list GET given empty list
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)

---
Failed   <<< connext_cmake_module [0.84s, exited with code 1]

ahmadnav, 2020-12-14:
Can be resolved by modifying:
/home/loacaladmin/ros2_android_ws/src/ros2/rosidl_typesupport_connext/connext_cmake_module/cmake/Modules/FindConnext.cmake

Comment out the libraries matching check in _find_connext_libraries  function

    # keep libraries matching this platform
    # if(match)
    #   math(EXPR i "${i} + 1")
    # else()
    #   list(REMOVE_AT libraries ${i})
    #   message("REMOVING FOUND LIBRARIES:::::::::")
    # endif()

This is more of a hack, a better option would be to find an older release of the rti connext library

esteve, 2020-12-14:
@ahmadnav RTI Connext is not supported for Android, could you try removing the `rosidl_typesupport_connext` package from your workspace and try again? Thanks.

esteve, 2020-12-14:
Also, how did you end up with `rosidl_typesupport_connext` in your ros2_android workspace? What repositories did you clone? `rosidl_typesupport_connext` is not part of https://github.com/ros2-java/ros2_java/blob/master/ros2_java_android.repos

ahmadnav, 2020-12-14:
@esteve Thanks! Looks like  https://github.com/ros2-java/ros2_java/blob/c3a2d7ad2ea7061a88914cadd744a5c85a6269b8/ros2_java_android.repos#L130 
Adds the package

esteve, 2020-12-14:
@ahmadnav can you try again with the updated workspace? Thanks.

ahmadnav, 2020-12-14:
Finished <<< rcljava [24.9s]

Summary: 94 packages finished [4min 27s]
  77 packages had stderr output: action_msgs ament_cmake ament_cmake_copyright ament_cmake_core ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_jars ament_cmake_export_jni_libraries ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_flake8 ament_cmake_gmock ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_lint_cmake ament_cmake_pep257 ament_cmake_pytest ament_cmake_python ament_cmake_ros ament_cmake_target_dependencies ament_cmake_test ament_cmake_uncrustify ament_cmake_xmllint ament_java_resources ament_lint_auto ament_lint_common builtin_interfaces cyclonedds_cmake_module fastcdr fastrtps fastrtps_cmake_module gmock_vendor gtest_vendor launch_testing_ament_cmake osrf_testing_tools_cpp poco_vendor python_cmake_module rcl rcl_interfaces rcl_logging_noop rcljava rcljava_common rcpputils rcutils rmw rmw_cyclonedds_cpp rmw_fastrtps_cpp rmw_fastrtps_dynamic_cpp rmw_fastrtps_shared_cpp rmw_implementation rmw_implementation_cmake rosidl_adapter rosidl_cmake rosidl_default_generators rosidl_default_runtime rosidl_generator_c rosidl_generator_cpp rosidl_generator_java rosidl_parser rosidl_typesupport_c rosidl_typesupport_cpp rosidl_typesupport_fastrtps_c rosidl_typesupport_fastrtps_cpp rosidl_typesupport_interface rosidl_typesupport_introspection_c rosidl_typesupport_introspection_cpp std_msgs test_interface_files test_msgs tinydir_vendor uncrustify_vendor unique_identifier_msgs

@esteve  @jacobperron Thanks for the quick response guys ROS2_java and android is compiling on system with [Ubuntu bionic, ROS Dashing, and NDK 17c(I am sure 21 will work too)]

ahmadnav, 2020-12-14:
#162 and #160 were the primary problems in this build.

closed 2020-12-14 by esteve