ros2-java/ros2_java #162: ros_android compile issue, on dashing branch with colcon build. Running UBUNTU 18.04 and ROS Dashing

ahmadnav, 2020-12-14:
Finished <<< rosidl_typesupport_cpp [0.39s]                                                                               
--- stderr: rosidl_generator_java                                
/opt/ndk/android-ndk-r21d/toolchains/llvm/prebuilt/linux-x86_64/lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: cannot find -lrosidl_generator_java__rosidl_typesupport_opensplice_c
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [rosidl_generator_java/rosidl_generator_java/msg/librosidl_generator_java_msg__nested__jni__rosidl_typesupport_opensplice_c.so] Error 1
make[1]: *** [CMakeFiles/msgNestedopensplice_c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/opt/ndk/android-ndk-r21d/toolchains/llvm/prebuilt/linux-x86_64/lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: cannot find -lrosidl_generator_java__rosidl_typesupport_opensplice_c
/opt/ndk/android-ndk-r21d/toolchains/llvm/prebuilt/linux-x86_64/lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: cannot find -lrosidl_generator_java__rosidl_typesupport_opensplice_c
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [rosidl_generator_java/rosidl_generator_java/srv/librosidl_generator_java_srv__basic_types__response__jni__rosidl_typesupport_opensplice_c.so] Error 1
make[1]: *** [CMakeFiles/srvBasicTypes_Responseopensplice_c.dir/all] Error 2
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [rosidl_generator_java/rosidl_generator_java/msg/librosidl_generator_java_msg_w_strings__jni__rosidl_typesupport_opensplice_c.so] Error 1
make[1]: *** [CMakeFiles/msgWStringsopensplice_c.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rosidl_generator_java [0.65s, exited with code 2]

ahmadnav, 2020-12-14:
Tried with NDK 21d and 17c

ahmadnav, 2020-12-14:
Removing /ros2/rmw_opensplice/ src/ros2/rosidl_typesupport_opensplice/ fixes this

esteve, 2020-12-14:
@ahmadnav in that case, can you confirm that after the latest changes in the dashing branch it works? Thanks.

ahmadnav, 2020-12-14:
@esteve  https://github.com/ros2-java/ros2_java/issues/160#issuecomment-744682564

PR #163 (2020-12-14): Remove dependency on OpenSplice
Removes both typesupport and rmw packages for OpenSplice for Android.

Fixes #162
closed 2020-12-14 by esteve