ros2-java/ros2_java #166: deprecated warnings in Android Studio

vkuehn, 2020-12-27:
@esteve  first of al sorry for bothering by mail and with this new issue !
there are some deprecated issues with an up to date Android Studio and the ros2_... examples
is there any discussion/work etc where this is dicussed or any hint how to work arround it ?

`* What went wrong:
A problem occurred configuring root project 'ros2_talker_android'.
> Failed to notify project evaluation listener.
   > org.gradle.api.tasks.compile.CompileOptions.setBootClasspath(Ljava/lang/String;)V

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2m 10s

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.1.1/userguide/command_line_interface.html#sec:command_line_warnings
`

ahmadnav, 2021-01-10:
I am assuming dashing and building with colon NOT ament

@vkuehn 
This error can be resolved by creating gradle/wrapper/gradle-wrapper.properties 
and writing distributionUrl=https\://services.gradle.org/distributions/gradle-4.9-all.zip to it

basically any Gradle less than v5.0
@esteve I assume making the ros_android_ws via ament automatically creates this file but on the dashing branch we use colcon

However, I get  Missing Ament Gradle property: ament.build_space error this too I assume is due to colcon vs ament

P.S.
 @esteve If we were to create our own separate app, and use the exact same files from the android examples (listener talker), and add the .so and .jar files from the install folder to the android app is expected to run? Because I keep getting the run time 

error org.ros2.rcljava.exceptions.RCLException: Failed to create node: type support not from this implementation, 

and I cannot resolve it.

vkuehn, 2021-01-11:
basically any Gradle less than v5.0-...not sure if that is really a solution...probably simply needs some upgrades. 
For that I am not using it at the moment. Feel free to close the issue

esteve, 2021-01-11:
@ahmadnav I kindly ask you not to hijack other tickets, whatever issues you have. If your goal is for us the maintainers to help you, it's much better for us, if you create separate tickets, instead of using other people's tickets. 

've tried helping you, but to be honest there's so much I can do, other people have been able to follow the instructions and get their projects running. Copy the generated .so and JAR files into your Android Studio project and you should be set.

vkuehn, 2021-01-11:
@esteve thanks. On the backend I am already on Foxy. I'll close this and keep watching. Will be a great time when all these mobiles become robots with a ros2 lts version
Thanks for the work

closed 2021-01-11 by vkuehn