ros2-java/ros2_java #167: Undefined symbol for custom defined message

xingjl6280, 2020-12-31:
Hi team,

I'm trying to define a new message in the directory ros2/example_interffaces/msg

VehicleStatus.msg：
*************************************************************
float64 x
float64 y
float64 z
*************************************************************

And modified CmakeLists.txt.
The part I modified:
*************************************************************
find_package(std_msgs REQUIRED)

set(msg_files
  "msg/VehicleStatus.msg"
)

rosidl_generate_interfaces(${PROJECT_NAME}
    ${msg_files}
    ${srv_files}
    DEPENDENCIES std_msgs
    ADD_LINTER_TESTS
)
*************************************************************

And modifed package.xml to add build and exec dependence on stat_msgs.

The colcon build works well. And I can reference the class in Java code.
But i got below runtime error:

2020-12-31 20:07:34.362 ERROR 11929 --- [           main] geometry_msgs.msg.VehicleStatus2         : Native code library failed to load.
java.lang.UnsatisfiedLinkError: /home/xingjl/ros2_java_ws/build/geometry_msgs/rosidl_generator_java/geometry_msgs/msg/libgeometry_msgs_msg__vehicle_status2__jni__rosidl_typesupport_c.so: /home/xingjl/ros2_java_ws/build/geometry_msgs/rosidl_generator_java/geometry_msgs/msg/libgeometry_msgs_msg__vehicle_status2__jni__rosidl_typesupport_c.so: undefined symbol: geometry_msgs__msg__VehicleStatus2__destroy
pure virtual method called
terminate called without an active exception

Please kindly advise if anything missing to be added. thank you

jacobperron, 2021-01-05:
Some questions to help clarify the issue:

Which version of ROS 2 are you using? Also, which version of rcljava?

The error message looks odd. You said you modified `example_interfaces`, but I see that the link error is related to `geometry_msgs`. Can you clarify the name of the package you are adding the message definition to and the name of the package that is using the generated Java code?

What kind of package are you building; a CMake package? a Gradle package?

Providing entire package files (e.g. CMakeLists.txt, package.xml, and source file) that reproduce the issue would be helpful.

xingjl6280, 2021-01-05:
Hi Jacob,

Thanks for your reply.

I'm using the dashing branch, latest version, for ros2 dashing. 

I've tried to add new msg in both gemetry_msgs and example_interfaces packages and successfully build them by colcon.
In a new gradle springboot project,  I tried to publish the new msgs into topic. 
The building of the gradle project completed without error, but with runtime error: not find the geometry_msgs__msg__VehicleStatus2__destroy method.

For already existing msgs, I can publish them with no issue.

From the runtime error, we can see the java class indeed got generated, and my code could find so lib file. 
But seems destroy method implementation not generated.

xingjl6280, 2021-01-05:
Just tried again. The error has gone away. 
Nothing changed but a reboot...

closed 2021-01-05 by xingjl6280