ros2-java/ros2_java #169: Client/Service example does not work

xingjl6280, 2021-01-06:
Hi team,

I tried to run the AddTwoInts service then client, but the client blocked at future.get().
I tried re-launch the client again, same issue.
Seems the service not registered properly, or the client could not find it somehow.

Please kindly advise how to make it work.

Btw, the pub/sub exmpale works well.

**Here's my setup:**
Ros2 version: dashing
RCLJava version: dashing branch checked out few days ago, build with colcon

**The service log:**
**ros_java_ws/build/rcljava_examples/build/classes/java/main$ java org.ros2.rcljava.examples.service.AddTwoIntsService**
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/home/xingjl/IdeaProjects/benchmark/ros_java_ws/build/rcljava_common/jars/slf4j-jdk14-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/home/xingjl/IdeaProjects/benchmark/ros_java_ws/build/rcljava_common/jars/slf4j-log4j12-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [org.slf4j.impl.JDK14LoggerFactory]
Jan 06, 2021 7:09:39 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_rcl_java__jni
Jan 06, 2021 7:09:39 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_contexts__context_impl__jni
Jan 06, 2021 7:09:39 PM org.ros2.rcljava.RCLJava rclJavaInit
INFO: Using RMW implementation: rmw_fastrtps_cpp
Jan 06, 2021 7:09:39 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_node__node_impl__jni
Jan 06, 2021 7:09:39 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_time__clock__jni
Jan 06, 2021 7:09:39 PM org.ros2.rcljava.common.JNIUtils loadTypesupport
INFO: Loading typesupport: example_interfaces_srv__add_two_ints__jni__rosidl_typesupport_c
Jan 06, 2021 7:09:39 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_service__service_impl__jni
Jan 06, 2021 7:09:39 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_executors__base_executor__jni

**The client log:
ros_java_ws/build/rcljava_examples/build/classes/java/main$ java org.ros2.rcljava.examples.client.AddTwoIntsClient**
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/home/xingjl/IdeaProjects/benchmark/ros_java_ws/build/rcljava_common/jars/slf4j-jdk14-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/home/xingjl/IdeaProjects/benchmark/ros_java_ws/build/rcljava_common/jars/slf4j-log4j12-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [org.slf4j.impl.JDK14LoggerFactory]
Jan 06, 2021 7:11:04 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_rcl_java__jni
Jan 06, 2021 7:11:04 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_contexts__context_impl__jni
Jan 06, 2021 7:11:04 PM org.ros2.rcljava.RCLJava rclJavaInit
INFO: Using RMW implementation: rmw_fastrtps_cpp
Jan 06, 2021 7:11:04 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_node__node_impl__jni
Jan 06, 2021 7:11:04 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_time__clock__jni
Jan 06, 2021 7:11:04 PM org.ros2.rcljava.common.JNIUtils loadTypesupport
INFO: Loading typesupport: example_interfaces_srv__add_two_ints__jni__rosidl_typesupport_c
Jan 06, 2021 7:11:04 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_client__client_impl__jni
Jan 06, 2021 7:11:04 PM org.ros2.rcljava.common.JNIUtils loadTypesupport
INFO: Loading typesupport: example_interfaces_srv__add_two_ints__request__jni__rosidl_typesupport_c
Jan 06, 2021 7:11:04 PM org.ros2.rcljava.common.JNIUtils loadImplementation
INFO: Loading implementation: org_ros2_rcljava_executors__base_executor__jni

xingjl6280, 2021-01-12:
Resolved. 
Needs to wait for the client fully up and find the service before making the request.

Created a PR.

jacobperron, 2021-01-14:
For reference, here is your PR https://github.com/ros2-java/ros2_java_examples/pull/8.

I think my existing PR also fixes this issue: https://github.com/ros2-java/ros2_java_examples/pull/6

@esteve Can you take a look at the referenced PRs? I believe you are the only person with write access to the examples repo.

xingjl6280, 2021-01-15:
You are right, Jacob, 
Sorry for raising a duplicated PR, I will withdraw it.

closed 2021-01-15 by xingjl6280

esteve, 2021-01-15:
@jacobperron I thought I had given you and @ivanpauno access to the repo, I've added you two as maintainers.