ros2-java/ros2_java #175: Errors when running ament build command

Arctic-Fox, 2021-02-09:
When performing the setup under "Desktop" I am seeing errors in the output:

arcticfox@gypsy-danger:~/Documents/workspace/ros2_java_ws$ ament build --symlink-install --isolated > buildlog.log
In file included from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_c/action_msgs/msg/goal_status__struct.h:18,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_c/action_msgs/msg/goal_status__functions.h:18,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_c/action_msgs/msg/goal_status__functions.c:4:
/home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_c/action_msgs/msg/goal_info__struct.h:18:10: fatal error: unique_identifier_msgs/msg/uuid__struct.h: No such file or directory
   18 | #include "unique_identifier_msgs/msg/uuid__struct.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/action_msgs__rosidl_generator_c.dir/build.make:171: CMakeFiles/action_msgs__rosidl_generator_c.dir/rosidl_generator_c/action_msgs/msg/goal_status__functions.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_c/action_msgs/msg/goal_info__functions.h:18,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_c/action_msgs/msg/goal_info__functions.c:4:
/home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_c/action_msgs/msg/goal_info__struct.h:18:10: fatal error: unique_identifier_msgs/msg/uuid__struct.h: No such file or directory
   18 | #include "unique_identifier_msgs/msg/uuid__struct.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/action_msgs__rosidl_generator_c.dir/build.make:158: CMakeFiles/action_msgs__rosidl_generator_c.dir/rosidl_generator_c/action_msgs/msg/goal_info__functions.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:549: CMakeFiles/action_msgs__rosidl_generator_c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /opt/ros/foxy/include/rosidl_runtime_cpp/message_initialization.hpp:18,
                 from /opt/ros/foxy/include/unique_identifier_msgs/msg/detail/uuid__struct.hpp:9,
                 from /opt/ros/foxy/include/unique_identifier_msgs/msg/uuid.hpp:7,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_info__struct.hpp:23,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_info.hpp:7,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status__struct.hpp:22,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_typesupport_introspection_cpp/action_msgs/msg/goal_status__type_support.cpp:12:
/opt/ros/foxy/include/rosidl_runtime_c/message_initialization.h:23:3: error: ‘ROSIDL_RUNTIME_C_MSG_INIT_ALL’ conflicts with a previous declaration
   23 |   ROSIDL_RUNTIME_C_MSG_INIT_ALL,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/arcticfox/Documents/workspace/ros2_java_ws/install_isolated/rosidl_generator_cpp/include/rosidl_generator_cpp/message_initialization.hpp:20,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status__struct.hpp:14,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_typesupport_introspection_cpp/action_msgs/msg/goal_status__type_support.cpp:12:
/home/arcticfox/Documents/workspace/ros2_java_ws/install_isolated/rosidl_generator_c/include/rosidl_generator_c/message_initialization.h:25:3: note: previous declaration ‘rosidl_runtime_c_message_initialization ROSIDL_RUNTIME_C_MSG_INIT_ALL’
   25 |   ROSIDL_RUNTIME_C_MSG_INIT_ALL,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/foxy/include/rosidl_runtime_cpp/message_initialization.hpp:18,
                 from /opt/ros/foxy/include/unique_identifier_msgs/msg/detail/uuid__struct.hpp:9,
                 from /opt/ros/foxy/include/unique_identifier_msgs/msg/uuid.hpp:7,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_info__struct.hpp:23,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_info.hpp:7,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status__struct.hpp:22,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_typesupport_introspection_cpp/action_msgs/msg/goal_status__type_support.cpp:12:
/opt/ros/foxy/include/rosidl_runtime_c/message_initialization.h:26:3: error: ‘ROSIDL_RUNTIME_C_MSG_INIT_SKIP’ conflicts with a previous declaration
   26 |   ROSIDL_RUNTIME_C_MSG_INIT_SKIP,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/arcticfox/Documents/workspace/ros2_java_ws/install_isolated/rosidl_generator_cpp/include/rosidl_generator_cpp/message_initialization.hpp:20,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status__struct.hpp:14,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_typesupport_introspection_cpp/action_msgs/msg/goal_status__type_support.cpp:12:
/home/arcticfox/Documents/workspace/ros2_java_ws/install_isolated/rosidl_generator_c/include/rosidl_generator_c/message_initialization.h:28:3: note: previous declaration ‘rosidl_runtime_c_message_initialization ROSIDL_RUNTIME_C_MSG_INIT_SKIP’
   28 |   ROSIDL_RUNTIME_C_MSG_INIT_SKIP,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/foxy/include/rosidl_runtime_cpp/message_initialization.hpp:18,
                 from /opt/ros/foxy/include/unique_identifier_msgs/msg/detail/uuid__struct.hpp:9,
                 from /opt/ros/foxy/include/unique_identifier_msgs/msg/uuid.hpp:7,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_info__struct.hpp:23,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_info.hpp:7,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status__struct.hpp:22,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_typesupport_introspection_cpp/action_msgs/msg/goal_status__type_support.cpp:12:
/opt/ros/foxy/include/rosidl_runtime_c/message_initialization.h:29:3: error: ‘ROSIDL_RUNTIME_C_MSG_INIT_ZERO’ conflicts with a previous declaration
   29 |   ROSIDL_RUNTIME_C_MSG_INIT_ZERO,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/arcticfox/Documents/workspace/ros2_java_ws/install_isolated/rosidl_generator_cpp/include/rosidl_generator_cpp/message_initialization.hpp:20,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status__struct.hpp:14,
                 from /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/rosidl_typesupport_introspection_cpp/action_msgs/msg/goal_status__type_support.cpp:12:
/home/arcticfox/Documents/workspace/ros2_java_ws/install_isolated/rosidl_generator_c/include/rosidl_generator_c/message_initialization.h:31:3: note: previous declaration ‘rosidl_runtime_c_message_initialization ROSIDL_RUNTIME_C_MSG_INIT_ZERO’
   31 |   ROSIDL_RUNTIME_C_MSG_INIT_ZERO,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/foxy/include/rosidl_runtime_cpp/message_initialization.hpp:18,
[... 271 lines cut ...]
[... cut ...]
<== Command '. /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs' with exit code '2'
<== Command '. /home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/arcticfox/Documents/workspace/ros2_java_ws/build_isolated/action_msgs' with exit code '2'
[buildlog.log](https://github.com/ros2-java/ros2_java/files/5952548/buildlog.log)

Arctic-Fox, 2021-02-13:
I had some source lines in my bashrc that I thought may be causing the issue so I commented them out and started over in a new terminal with a new folder.  Started from the ament_ws setup and worked my way through.  Once again, there are errors on the last line of the ros2 setup Desktop section:

CMake Error at CMakeLists.txt:16 (find_package):
  By not providing "Findunique_identifier_msgs.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "unique_identifier_msgs", but CMake did not find one.

  Could not find a package configuration file provided by
  "unique_identifier_msgs" with any of the following names:

    unique_identifier_msgsConfig.cmake
    unique_identifier_msgs-config.cmake

  Add the installation prefix of "unique_identifier_msgs" to
  CMAKE_PREFIX_PATH or set "unique_identifier_msgs_DIR" to a directory
  containing one of the above files.  If "unique_identifier_msgs" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/arcticfox/workspaces/ros2_java_ws/build_isolated/action_msgs/CMakeFiles/CMakeOutput.log".

<== Command '. /home/arcticfox/workspaces/ros2_java_ws/build_isolated/action_msgs/cmake__build.sh && /usr/bin/cmake -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/arcticfox/workspaces/ros2_java_ws/install_isolated/action_msgs /home/arcticfox/workspaces/ros2_java_ws/src/ros2/rcl_interfaces/action_msgs' failed in '/home/arcticfox/workspaces/ros2_java_ws/build_isolated/action_msgs' with exit code '1'
<== Command '. /home/arcticfox/workspaces/ros2_java_ws/build_isolated/action_msgs/cmake__build.sh && /usr/bin/cmake -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/arcticfox/workspaces/ros2_java_ws/install_isolated/action_msgs /home/arcticfox/workspaces/ros2_java_ws/src/ros2/rcl_interfaces/action_msgs' failed in '/home/arcticfox/workspaces/ros2_java_ws/build_isolated/action_msgs' with exit code '1'
[CMakeOutput.log](https://github.com/ros2-java/ros2_java/files/5976197/CMakeOutput.log)

rs-silva, 2021-02-21:
> I had some source lines in my bashrc that I thought may be causing the issue so I commented them out and started over in a new terminal with a new folder. Started from the ament_ws setup and worked [...]

I've this exact problem and can't fix it either, I don't understand what's wrong

rs-silva, 2021-02-22:
@Arctic-Fox I was able to fix the error by installing Dashing distribution (I was using Bouncy before).

However, I encountered other errors, also in action_msgs

Arctic-Fox, 2021-02-22:
> @Arctic-Fox I was able to fix the error by installing Dashing distribution (I was using Bouncy before).
> 
> However, I encountered other errors, also in action_msgs

@re-silva I'm still stuck in the same place with the action_msgs.  Will advise if I find a solution.

jacobperron, 2021-11-13:
The recommended approach is to use the `main` branch, which will target the newest ROS distro (currently Galactic).
The README has been updated and hopefully the instructions work (please open an issue if you find this is not the case).

Note, you should no longer use `ament build`; instead `colcon` is the preferred build tool.

closed 2021-11-13 by jacobperron