ros2-java/ros2_java #178: Support for ROS2 Rolling

gc-robotics, 2021-05-12:
Hi,

Does ros2_java build on Rolling, Foxy, etc? 
I have seen the foxy-devel branch but not looking great to me...

Regards.

jacobperron, 2021-05-12:
We have been staging work for Foxy and Galactic support over on this fork: https://github.com/osrf/ros2_java
The plan is to start getting those changes upstreamed here. Stay tuned!

mclayton7, 2021-06-18:
@jacobperron I'm interested in using this with ROS2 Galactic. Is there a place you guys are tracking all of the work left for Foxy/Galactic support? Thanks!

jacobperron, 2021-06-19:
@mclayton7 I've been coordinating with @ivanpauno, who has starting porting changes from our fork here (E.g.  https://github.com/ros2-java/ros2_java/pull/186). We are currently using that fork with Galactic. The plan moving forward is that ros2_java will target the latest rosdistro. Therefore, we won't guarantee support for Foxy, though there may be some previous commits that are compatible.

mclayton7, 2021-06-22:
Thanks @jacobperron! If I'm interested in using the Galactic/Rolling version, should I try to get osrf/ros2_java working  for me, or help get it merged in to ros2-java/ros2_java?

jacobperron, 2021-06-29:
> should I try to get osrf/ros2_java working for me, or help get it merged in to ros2-java/ros2_java?

Both sound good to me :smile:

quyenanh, 2021-08-17:
> Thanks @jacobperron! If I'm interested in using the Galactic/Rolling version, should I try to get osrf/ros2_java working for me, or help get it merged in to ros2-java/ros2_java?
Does it work?

jacobperron, 2021-11-13:
To follow-up here, the `main` branch of this repo should now work with ROS Galactic.

Due to resource constraints, our plans going forward are to support the latest ROS distribution. So, we won't support Rolling; any updates will likely be made around or after the ROS Humble is release date.

I'm going to close this since it is more of a question. Feel free to continue commenting.

closed 2021-11-13 by jacobperron