ros2-java/ros2_java #194: Compile error with main branch, Ubuntu focal

ahmadnav, 2021-11-14:
Finished <<< ament_cmake_export_jni_libraries [0.15s]                                                                                                                                                        
Finished <<< ament_cmake_export_jars [0.16s]
--- stderr: rcutils                                                                                                 
CMake Error at CMakeLists.txt:187 (find_package):
  By not providing "Findmimick_vendor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "mimick_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "mimick_vendor"
  with any of the following names:

    mimick_vendorConfig.cmake
    mimick_vendor-config.cmake

  Add the installation prefix of "mimick_vendor" to CMAKE_PREFIX_PATH or set
  "mimick_vendor_DIR" to a directory containing one of the above files.  If
  "mimick_vendor" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< rcutils [0.42s, exited with code 1]
Aborted  <<< rosidl_parser [0.45s]
Aborted  <<< fastrtps [2.73s]                                  

Summary: 69 packages finished [9.13s]
  1 package failed: rcutils
  2 packages aborted: fastrtps rosidl_parser
  2 packages had stderr output: fastrtps rcutils
  34 packages not processed

ahmadnav, 2021-11-14:
I have double checked that the package mimick_vendor has been installed, I have tried 
- Installing it by itself
- Setting the mimick_vendor_DIR path my self to the path of mimick_vendorConfig.cmake

ahmadnav, 2021-11-15:
This error can be resolved by 
  set(mimick_vendor_DIR /home/ros2_android_dep/install/mimick_vendor/share/mimick_vendor/cmake)
  set(mimick_DIR /home/ros2_android_dep/install/mimick_vendor/lib/cmake/mimick/)

jacobperron, 2021-11-15:
I presume you are following the build instructions for Android, is this correct?

We worked around the issue with mimick in our CI by not building tests for some packages. This is done by creating a colcon.meta file in the workspace with the following contents:

https://github.com/ros2-java/ros2_java/blob/5134c544c16ca4045d9da5915fdf38f9d2410c3f/.github/workflows/build_and_test.yml#L37-L61

I think we should update the README to reflect the changes in the CI script.

ahmadnav, 2021-11-15:
> I presume you are following the build instructions for Android, is this correct?
>
> We worked around the issue with mimick in our CI by not building tests for some packages. This is done by creating a [...]

That's true. Thanks Jacob I would also like to add the you have to add -DBOOST_ROOT=/opt/boost/boost_1_77_0/boost 
other wise asio.hpp is not detected.

ahmadnav, 2021-11-15:
I have gotten the same issue with spdlog_vendor, will update when I find the solution. After setting -Dspdlog_vendor_DIR=/opt/ros/galactic/share/spdlog_vendor/cmake the error than changes to 
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.

I have no solution for this at the moment as there is no such files as     spdlogConfig.cmake or   spdlog-config.cmake

jacobperron, 2021-11-16:
We shouldn't need spdlog as I'm pretty sure we're not relying on `rcl_logging`.

I'll try going through the Android setup myself and see if I can update the instructions for a smoother experience.

ahmadnav, 2021-11-17:
> We shouldn't need spdlog as I'm pretty sure we're not relying on `rcl_logging`.
> 
> I'll try going through the Android setup myself and see if I can update the instructions for a smoother experience.

Thanks Jacob, Now I get the following error;

eProsima/Fast-DDS/src/cpp/rtps/transport/UDPChannelResource.h:18:10: fatal error: 'asio.hpp' file not found

I definitely have boost and libiso installed, and can see asio.hpp in /usr/include.

ahmadnav, 2021-11-26:
@jacobperron 

--- stderr: rcl                              
CMake Error at cmake/get_default_rcl_logging_implementation.cmake:43 (find_package):
  By not providing "Findrcl_logging_spdlog.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "rcl_logging_spdlog", but CMake did not find one.

  Could not find a package configuration file provided by
  "rcl_logging_spdlog" with any of the following names:

    rcl_logging_spdlogConfig.cmake
    rcl_logging_spdlog-config.cmake

  Add the installation prefix of "rcl_logging_spdlog" to CMAKE_PREFIX_PATH or
  set "rcl_logging_spdlog_DIR" to a directory containing one of the above
  files.  If "rcl_logging_spdlog" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:18 (get_default_rcl_logging_implementation)

---
Failed   <<< rcl [0.47s, exited with code 1]

Looks like SPDLog is necessary 

###############

/home//ros2_android_ws/src/eProsima/Fast-DDS/src/cpp/utils/IPFinder.cpp:165:9: error: use of undeclared identifier 'getifaddrs'; did you mean 'ifaddrs'?
    if (getifaddrs(&ifaddr) == -1)
        ^
/opt/ndk/android-ndk-r21e/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/ifaddrs.h:45:8: note: 'ifaddrs' declared here
struct ifaddrs {
       ^
/home//ros2_android_ws/src/eProsima/Fast-DDS/src/cpp/utils/IPFinder.cpp:223:5: error: use of undeclared identifier 'freeifaddrs'
    freeifaddrs(ifaddr);
    ^

Solution:
ANDROID_NDK=/opt/ndk/android-ndk-r21e 
Set API >=24

ahmadnav, 2021-11-26:
As for the asio.hpp issue you need to git submodule update --init --recursive within the Fast-DDS directory, to get the asio libraries under thirdparty (which are currently link folders so aren't cloned) than add 

        $<BUILD_INTERFACE:${PROJECT_SOURCE_DIR}/thirdparty/asio/asio/include>
        $<BUILD_INTERFACE:${PROJECT_SOURCE_DIR}/thirdparty/tinyxml2>

within target_include_directories(${PROJECT_NAME} PUBLIC )

ahmadnav, 2021-11-26:
@jacobperron I can't seem to find a solution to the spd log issue. Not sure why spdlog_vendor requires spdlog as well

huyubao, 2021-12-17:
I also encountered this problem when compiling Android. How do you solve it, mimick_vendor How to install 

CMake Error at CMakeLists.txt:187 (find_package):
  By not providing "Findmimick_vendor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "mimick_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "mimick_vendor"
  with any of the following names:

    mimick_vendorConfig.cmake
    mimick_vendor-config.cmake

  Add the installation prefix of "mimick_vendor" to CMAKE_PREFIX_PATH or set
  "mimick_vendor_DIR" to a directory containing one of the above files.  If
  "mimick_vendor" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< rcutils [0.97s, exited with code 1]
Aborted  <<< rosidl_runtime_cpp [5.46s]                  
Aborted  <<< fastrtps [48.0s]

adarshjamuaar, 2024-04-03:
Solved it by cloning https://github.com/ros2/mimick_vendor into the build folder.
facing some other ndk undefined symbol errors as of now :) when it builds the google_benchmark_vendor