ros2-java/ros2_java #197: Miss support for security plugins in ros2 like subscriber/publisher.

piaopiaohu123, 2021-11-24:
hi gus:
     
     Found three problems or topics we need improve:
        1. ros2_java miss support for security plugins in ros2
        2.and when we need to add security plugins ,  we need build openssl lib on arm64 v8 or v7 depends version of android. 
        3.people from china can`t visit many repo like maven service in USA, need a special doc for them.
     i`m already so some works on these, need options from you, do we have any plan in future?

jacobperron, 2021-12-01:
Hi @piaopiaohu123, thanks for opening this issue. I don't think we have any plans related to security and ROS 2 Java, any help is appreciated.

To help clarify, are you referring to using [sros2](https://github.com/ros2/sros2) to take advantage of the middleware security?
If so, what specifically is preventing this?

Or is there something else you had in mind?

piaopiaohu123, 2021-12-16:
> Hi @piaopiaohu123, thanks for opening this issue. I don't think we have any plans related to security and ROS 2 Java, any help is appreciated.
>
> To help clarify, are you referring to using [...]

 security is all set by dds security middleware there, but miss doc for all the step, you`re right, don`t need to do anything here.

i`m want to maintain this repo in future, can you add me into your group? i can do some thing just like solve the issue and problems.

piaopiaohu123, 2021-12-16:
> Hi @piaopiaohu123, thanks for opening this issue. I don't think we have any plans related to security and ROS 2 Java, any help is appreciated.
>
> To help clarify, are you referring to using [...]

when build ros2 and open macro SECURITY=ON, it depent openssl on arm64 , but i don`t find any doc or example to suppurt that, so wonder where it should be added, here  or  ros2 repo?

jacobperron, 2022-01-06:
@pengzhouhu I think any documentation related to enabling security can be added to this repository. I'm happy to review a pull request if you have made some progress related to this.

I think any example code can be added to https://github.com/ros2-java/ros2_java_examples or https://github.com/ros2-java/ros2_android_examples

pengzhouhu, 2022-01-07:
> @pengzhouhu I think any documentation related to enabling security can be added to this repository. I'm happy to review a pull request if you have made some progress related to this.
>
> I think any [...]

@jacobperron   OK，i will handle that.