ros2-java/ros2_java #198: MultiThreadedExecutor does not process callbacks asynchronously

jacobperron, 2021-12-01:
Each callback is guarded by mutex, so even though they are happening in different threads, they are processed in series:

https://github.com/ros2-java/ros2_java/blob/ae5edaa0f07f14938913bb5e395544e3ca4d91c5/rcljava/src/main/java/org/ros2/rcljava/executors/MultiThreadedExecutor.java#L85-L87

I proposed a patch to actually execute callbacks in parallel by default (https://github.com/osrf/ros2_java/pull/86), but @ivanpauno pointed out that it would be more consistent with other client libraries to introduce the concept of callback groups. I agree with this as it would not only be consistent, but make it more flexible to configure the executor.

scpeters, 2021-12-08:
Is there a ticket yet about supporting callback groups?

jacobperron, 2022-01-06:
> Is there a ticket yet about supporting callback groups?

That's a negative.