ros2-java/ros2_java #201: Build ROS 2 Java for Android failed

xcpper, 2021-12-22:
Ubuntu 20.04.3 && ROS2 Galactic && openjdk8 && Android Studio

Following README.md
1: Download NDK(16.1.4479499) and set the environment variable ANDROID_NDK
2: Download SDK(API level 21) and set the environment variable ANDROID_HOME
3: Clone ROS 2 and ROS 2 Java for Android source code
4: Set Android build configuration
5: Build 

But colcon build failed: 
……/Android/Sdk/ndk/16.1.4479499/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: cannot find -lc++_shared

piaopiaohu123, 2021-12-22:
@xcpper i have ever meet this problem, need more information of 4 step: Android build configuration

xcpper, 2021-12-22:
@pengzhouhu I did not change any command in README.md

 export PYTHON3_EXEC="$( which python3 )"
 export PYTHON3_LIBRARY="$( ${PYTHON3_EXEC} -c 'import os.path; from distutils import sysconfig; print(os.path.realpath(os.path.join(sysconfig.get_config_var("LIBPL"), sysconfig.get_config_var("LDLIBRARY"))))' )"
 export PYTHON3_INCLUDE_DIR="$( ${PYTHON3_EXEC} -c 'from distutils import sysconfig; print(sysconfig.get_config_var("INCLUDEPY"))' )"
 export ANDROID_ABI=armeabi-v7a
 export ANDROID_NATIVE_API_LEVEL=android-21
 export ANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang

pengzhouhu, 2021-12-23:
> @pengzhouhu I did not change any command in README.md
>
> export PYTHON3_EXEC="$( which python3 )" export PYTHON3_LIBRARY="$( ${PYTHON3_EXEC} -c 'import os.path; from distutils import sysconfig; [...]

i remember there is a so named  "lib_c++_shared.so"? maybe you miss copy it where build the rclandroid.aar.

samcts2309, 2022-03-09:
> i remember there is a so named "lib_c++_shared.so"? maybe you miss copy it where build the rclandroid.aar.

May i know where does the rclandroid.aar build to?
I tried to find it in the install folder or build folder, its no where to be found after `colcon build`