ros2-java/ros2_java #202: nee to improve security document in rcljava android examples.

pengzhouhu, 2021-12-27:
@jacobperron   hi jacob, i take the issue here:https://github.com/ros2-java/ros2_java/issues/197,    my real mind is that when we need add security model through rcljava android examples, like set some security args, we miss a document to discribe how people can do it. 

the example miss this case, do we need to improve it ?

jacobperron, 2022-01-06:
If you are able to contribute documentation for how to enable security for rcljava on Android, I'm happy to review a pull request :)

pengzhouhu, 2022-01-07:
> If you are able to contribute documentation for how to enable security for rcljava on Android, I'm happy to review a pull request :)

no problem，i will take it as my homework and push as soon.