ros2-java/ros2_java #218: how to build ros2 java on ros2-humble branch

kccccccccccccccccc, 2022-06-10:
now build error
info:
CMake Deprecation Warning at /Users/kang/code/ros2_humble/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  /Users/kang/code/ros2_humble/install/rosidl_generator_java/share/rosidl_generator_java/cmake/rosidl_generator_java_generate_interfaces.cmake:259 (rosidl_target_interfaces)
  /Users/kang/code/ros2_humble/install/ament_cmake_core/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /Users/kang/code/ros2_humble/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:16 (rosidl_generate_interfaces)

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/Users/kang/code/ros2_humble/src/ros2/rcl_interfaces/builtin_interfaces/include_directories
   used as include directory in directory /Users/kang/code/ros2_humble/src/ros2/rcl_interfaces/builtin_interfaces
   used as include directory in directory /Users/kang/code/ros2_humble/src/ros2/rcl_interfaces/builtin_interfaces
   used as include directory in directory /Users/kang/code/ros2_humble/src/ros2/rcl_interfaces/builtin_interfaces
   used as include directory in directory /Users/kang/code/ros2_humble/src/ros2/rcl_interfaces/builtin_interfaces
   used as include directory in directory /Users/kang/code/ros2_humble/src/ros2/rcl_interfaces/builtin_interfaces
   used as include directory in directory /Users/kang/code/ros2_humble/src/ros2/rcl_interfaces/builtin_interfaces

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< builtin_interfaces [1.72s, exited with code 1]
Aborted  <<< unique_identifier_msgs [5.35s]                                                                                  
Aborted  <<< rmw_dds_common [21.2s]

kccccccccccccccccc, 2022-06-10:
build script
colcon build \
--packages-ignore cyclonedds rcl_logging_log4cxx rosidl_generator_py \
--packages-up-to rcljava \
--metas colcon.meta \
--cmake-args \
-DPYTHON_EXECUTABLE=${PYTHON3_EXEC} \
-DPYTHON_LIBRARY=${PYTHON3_LIBRARY} \
-DPYTHON_INCLUDE_DIR=${PYTHON3_INCLUDE_DIR} \
-DCMAKE_TOOLCHAIN_FILE=${ANDROID_NDK}/build/cmake/android.toolchain.cmake \
-DANDROID_FUNCTION_LEVEL_LINKING=OFF \
-DANDROID_NATIVE_API_LEVEL=${ANDROID_NATIVE_API_LEVEL} \
-DANDROID_TOOLCHAIN_NAME=${ANDROID_TOOLCHAIN_NAME} \
-DANDROID_STL=c++_shared \
-DANDROID_ABI=${ANDROID_ABI} \
-DANDROID_NDK=${ANDROID_NDK} \
-DTHIRDPARTY=ON \
-DCOMPILE_EXAMPLES=OFF \
-DCMAKE_FIND_ROOT_PATH="${PWD}/install"

genevanmeter, 2022-08-04:
https://github.com/genevanmeter/ros2_java/tree/humble

My fork builds on Ubuntu 22.04 for humble.  I don't have experience with Java but these changes were to pull humble branches and a cmake change.

To use the build steps need to modified to use the different repos:
```curl -skL https://raw.githubusercontent.com/genevanmeter/ros2_java/humble/ros2_java_desktop.repos | vcs import src```

kccccccccccccccccc, 2022-08-18:
> https://github.com/genevanmeter/ros2_java/tree/humble
>
> My fork builds on Ubuntu 22.04 for humble. I don't have experience with Java but these changes were to pull humble branches and a cmake [...]

thanks , I will have a try

genevanmeter, 2022-08-23:
I was only concerned with getting it to run on desktop.

closed 2024-05-29 by kccccccccccccccccc