ros2-java/ros2_java #225: Android API level 30 (might not be an issue with this repo)

jtabraham123, 2022-09-29:
Hi, I am currently building an app that targets android api 30+ with ros2_java. However, for some reason I cannot get ros2 talker and listener nodes to communicate in api 30. It works in api 29 (I have copied example from https://github.com/YasuChiba/ros2-android-test-app). The talker and listener node can communicate locally on the app, but cannot talk to other devices with nodes on the same network? Has anyone else had this issue with api 30 and know a workaround?

jtabraham123, 2022-10-24:
Note: One thing ive noticed when I run api 30+ (Android 11) vs 29 and below, the node handles and context handles are now all negative. They were positive in 29 and below. This would not be the issue if everything is cast properly though. I am just wondering if it could be a casting thing.

jtabraham123, 2022-11-01:
I have a few ideas as to why this is happening, could be the middleware (in which case this issue would be better suited for one of those repos) or the socket layer.

YasuChiba, 2022-11-02:
I'm wondering if this is the cause. (Though I have not tried yet)

https://github.com/eProsima/Fast-DDS/pull/2677

jtabraham123, 2022-11-02:
Yes it is Thank you YasuChiba so much for pointing this out! Just had to update the repos file to Fast-DDS 2.8.0 (version with this commited) and Fast CDR to v1.0.25.

closed 2022-11-02 by jtabraham123