ros2-java/ros2_java #231: Build desktop packages:

Etoilesforever, 2023-04-07:
CMake Error at CMakeLists.txt:85 (string_camel_case_to_lower_case_underscore):
  Unknown CMake command "string_camel_case_to_lower_case_underscore".

joshuakimrub, 2023-06-26:
I experienced the same issue when I used ros2 iron. My solution was to use ros2 humbe instead.

ChristophSchwanke, 2025-01-22:
I found this solution:

```
index adf0512..e55a567 100644
--- a/rcljava/CMakeLists.txt
+++ b/rcljava/CMakeLists.txt
@@ -14,6 +14,7 @@ find_package(rmw REQUIRED)
 find_package(rmw_implementation_cmake REQUIRED)
 
 include(CrossCompilingExtra)
+include(/opt/ros/jazzy/share/rosidl_cmake/cmake/string_camel_case_to_lower_case_underscore.cmake)
 
 if(CMAKE_CROSSCOMPILING)
   find_host_package(Java COMPONENTS Development REQUIRED)
```