ros2-java/ros2_java #232: build on windows, rosidl_generator_java package failed

pengg, 2023-05-09:
Finished <<< ament_build_type_gradle [1.56s]
Starting >>> ament_cmake_export_jars
Starting >>> ament_cmake_export_jni_libraries
Finished <<< ament_cmake_export_jars [1.23s]
Starting >>> rcljava_common
Finished <<< ament_cmake_export_jni_libraries [1.31s]
Finished <<< rcljava_common [2.14s]
Starting >>> rosidl_generator_java
--- stderr: rosidl_generator_java
CMake Error at CMakeLists.txt:73 (get_property):
  get_property could not find TARGET rosidl_generator_java_messages_jar.
  Perhaps it has not yet been created.

CMake Error at E:/tools/ros2-humble/ros2-windows/src/ros2_java_galactic/install/share/rcljava_common/cmake/Modules/JavaExtra.cmake:104 (ament_export_jars):
  Unknown CMake command "ament_export_jars".
Call Stack (most recent call first):
  CMakeLists.txt:83 (ament_add_junit_tests)

---
Failed   <<< rosidl_generator_java [4.47s, exited with code 1]

Summary: 5 packages finished [11.3s]
  1 package failed: rosidl_generator_java
  2 packages had stderr output: ament_build_type_gradle rosidl_generator_java
  27 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)

gatordevin, 2024-01-18:
Did you ever resolve this issue?

adammetcalf, 2024-06-21:
Similar issue for me, but building on Ubuntu 20.