ros2-java/ros2_java #233: Service doesn't work on Android App

MarkJoson, 2023-05-29:
Hello, I have been trying to use service on my app, but seem to be running into some issues. I have seen the test code of service client in ros2_java/jcljava/src/test/java/org/ros2/rcljava/client/ClientTest.java, which also includes the example of creating a service server. I have tried to use createService like this code. However, the TestClientConsumer cannot be executed when I use ros2 tool to call this service on the other computer. Meanwhile, I tried to send request by the button on the gui of my android app, it could work sometime.

mathankumar-dotworld, 2024-07-07:
@MarkJoson is this fixed? . Am also facing same

MarkJoson, 2024-07-08:
I can't remember what I did, but the ros2 service seemed work well after some tries.