ros2-java/ros2_java #241: stderr: fastrtps

xuecq236, 2024-06-20:
stderr: fastrtps
CMake Error at CMakeLists.txt:231 (find_package):
  By not providing "Findfoonathan_memory.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "foonathan_memory", but CMake did not find one.

  Could not find a package configuration file provided by "foonathan_memory"
  with any of the following names:

    foonathan_memoryConfig.cmake
    foonathan_memory-config.cmake

  Add the installation prefix of "foonathan_memory" to CMAKE_PREFIX_PATH or
  set "foonathan_memory_DIR" to a directory containing one of the above
  files.  If "foonathan_memory" provides a separate development package or
  SDK, be sure it has been installed.

ubuntu 22.04
ndk 21e
android api 26

colcon build for android ,How should I solve it
[image]

xuecq236, 2024-06-24:
Solved, can be closed

huangjie84, 2024-07-11:
> Solved, can be closed

Hello,can u tell me how do u solve this ? I got same one.