ros2-java/ros2_java #243: "colcon build" failing error

guneer491, 2024-07-17:
‘Colcon build’ failing error

When I try to build ros2-java workspace, I face an issue in building ‘rclJava_examples’ pkg. Here’s the full log of error msg :-

Starting a Gradle Daemon (subsequent builds will be faster)

FAILURE: Build failed with an exception.

* What went wrong:
Unable to start the daemon process.
This problem might be caused by incorrect configuration of the daemon.
For example, an unrecognized jvm option is used.
Please refer to the user guide chapter on the daemon at https://docs.gradle.org/4.9/userguide/gradle_daemon.html
Please read the following process output to find out more:
-----------------------

FAILURE: Build failed with an exception.

* What went wrong:
java.lang.ExceptionInInitializerError (no error message)

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

* Try:
Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Exception is:
org.gradle.api.GradleException: Unable to start the daemon process.
This problem might be caused by incorrect configuration of the daemon.
For example, an unrecognized jvm option is used.
Please refer to the user guide chapter on the daemon at https://docs.gradle.org/4.9/userguide/gradle_daemon.html
Please read the following process output to find out more:
-----------------------

FAILURE: Build failed with an exception.

* What went wrong:
java.lang.ExceptionInInitializerError (no error message)

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

        at org.gradle.launcher.daemon.client.DaemonGreeter.parseDaemonOutput(DaemonGreeter.java:35)
        at org.gradle.launcher.daemon.client.DefaultDaemonStarter.startProcess(DefaultDaemonStarter.java:160)
        at org.gradle.launcher.daemon.client.DefaultDaemonStarter.startDaemon(DefaultDaemonStarter.java:136)
        at org.gradle.launcher.daemon.client.DefaultDaemonConnector.doStartDaemon(DefaultDaemonConnector.java:212)
        at org.gradle.launcher.daemon.client.DefaultDaemonConnector.startDaemon(DefaultDaemonConnector.java:206)
        at org.gradle.launcher.daemon.client.DefaultDaemonConnector.connect(DefaultDaemonConnector.java:130)
        at org.gradle.launcher.daemon.client.DaemonClient.execute(DaemonClient.java:139)
        at org.gradle.launcher.daemon.client.DaemonClient.execute(DaemonClient.java:93)
        at org.gradle.launcher.cli.RunBuildAction.run(RunBuildAction.java:52)
        at org.gradle.internal.Actions$RunnableActionAdapter.execute(Actions.java:207)
        at org.gradle.launcher.cli.CommandLineActionFactory$ParseAndBuildAction.execute(CommandLineActionFactory.java:402)
        at org.gradle.launcher.cli.CommandLineActionFactory$ParseAndBuildAction.execute(CommandLineActionFactory.java:375)
        at org.gradle.launcher.cli.ExceptionReportingAction.execute(ExceptionReportingAction.java:37)
        at org.gradle.launcher.cli.ExceptionReportingAction.execute(ExceptionReportingAction.java:23)
        at org.gradle.launcher.cli.CommandLineActionFactory$WithLogging.execute(CommandLineActionFactory.java:368)
        at org.gradle.launcher.cli.CommandLineActionFactory$WithLogging.execute(CommandLineActionFactory.java:298)
        at org.gradle.launcher.Main.doAction(Main.java:36)
        at org.gradle.launcher.bootstrap.EntryPoint.run(EntryPoint.java:45)
        at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:104)
        at java.base/java.lang.reflect.Method.invoke(Method.java:578)
        at org.gradle.launcher.bootstrap.ProcessBootstrap.runNoExit(ProcessBootstrap.java:60)
        at org.gradle.launcher.bootstrap.ProcessBootstrap.run(ProcessBootstrap.java:37)
        at org.gradle.launcher.GradleMain.main(GradleMain.java:23)

* Get more help at https://help.gradle.org
---
Failed   <<< rcljava_examples [0.82s, exited with code 1]

My environment :- 

Java 20.0.2
Ubuntu 22.04.3 LTS
ROS2 Humble
Gradle 4.9

Can anyone please provide some guidance on what could be the issue and how to resolve it. I can provide more information if needed. Thanks.

guneer491, 2024-07-22:
I was able to resolve this issue. Problem with java version

closed 2024-07-22 by guneer491

trillionY, 2024-08-21:
> I was able to resolve this issue. Problem with java version

Hi, may I ask what Java version did you use that resolve the problem? and did you use Gradle 4.9 ?